NebraLtd/AnyBeam #3: Adding CAD files for HAT?

beniroquai, 2020-10-22:
Today we received the Anybeam HAT. Finally! Nice! And it works. 

I was wondering if you can provide a 3D CAD model (e.g. STL/STP) for the assembly so that we can start building a case for it. 

Thanks!

ryanteck, 2020-10-22:
Hi @beniroquai ,

We'll look into it and see if we can generate one.

shawaj, 2020-10-23:
@beniroquai  We should have these today for you.

beniroquai, 2020-10-27:
@shawaj any update? :)

ryanteck, 2020-10-27:
We should have an update shortly, I believe we've got 2 of them generated now and are working on the last.

shawaj, 2020-11-08:
Closed by https://github.com/NebraLtd/AnyBeam/commit/5d639666c64d2efbbc10fafe64d329aa01558935

shawaj, 2020-11-08:
3D files are here - https://github.com/NebraLtd/AnyBeam/tree/master/Hardware/3D

closed 2020-11-08 by shawaj

shawaj, 2020-11-23:
@beniroquai it is not a priority for us at the moment, so apologies but we will not be able to do this