NebraLtd/AnyBeam #4: Projector stuck showing "Nebra Anybeam"

vpetersson, 2020-10-22:
I'm unsure if this is a hardware or software error, but I'm only getting the projector to display the splash screen.

I haven't tested with anything other than Kodi, but at least with Kodi this is the behavior. I've verified that it at least works with regular HDMI.

shawaj, 2020-10-22:
For setting up the AnyBeam HAT you have to configure it as a DPI display in the config.txt file.

We have a user guide here which explains the process: https://github.com/NebraLtd/AnyBeam/tree/master/Documentation/HAT

And it is also available in the software readme here: https://github.com/NebraLtd/AnyBeam/tree/master/Software

Hopefully that should allow it to work. If not then it could be a hardware issue, or possibly an issue with display selection in Kodi.

Let us know how you get on and we can debug other issues as they come up 👍

vpetersson, 2020-10-22:
Working on it but the `curl` installer won't work on Kodi due to the dependency on `apt` and `whiptail`. Need to read the script to see how to work around that :)

vpetersson, 2020-10-22:
That did the trick. Just had to manually append the following to `config.txt`:

```
#AnyBeam
dtoverlay=dpi24
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
framebuffer_width=1280
framebuffer_height=720
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=85
dpi_output_format=0x070027
```

closed 2020-10-22 by vpetersson

shawaj, 2020-10-22:
Awesome glad it's working :-)

Will make the docs more easily understandable for future batches

ryanteck, 2020-11-24:
@shawaj I've had another customer report issues still so going to confirm this works when I get my unit.

shawaj, 2020-11-24:
@ryanteck did Jacqui send you one?

ryanteck, 2020-11-24:
Should get it soon I believe.Chris was sending me one of his.

I'll try with the above manual and check, and if it works then upgrade the guide to have the details.

jhaip, 2020-11-25:
I ran into this same issue of only getting the projector to display the splash screen on my RPi4 while using the AnyBeam HAT. It didn't happen every time, but it happened 9 out of 10 times. I had added the settings from the manual to `config.txt`.

A fix that seems to work for me is also disabling the splash screen and commenting out the i2c and spi dtparam lines in my `/boot/config.txt`:

```
disable_splash=1
# dtparam=i2c_arm=on
# dtparam=i2s=on
# dtparam=spi=on
```

SinanAkkoyun, 2023-04-26:
Hey, I've tried everything above and even switched Pi's and still only the default screen!

I bought my hat a couple of weeks ago from eBay, I could simply not find anything due to it being discontinued.

Can anybody please help me out?