NebraLtd/AnyBeam #5: adjustable color intensity?

MoorPoor, 2020-10-26:
hei!
I`m using the RaspiHAT.
The projector has a strong red dominance. projecting white turns out very pinkish
Is there a way to regulate the intensity of the individual colors?
Thanks!

shawaj, 2020-10-26:
@MoorPoor could you share a picture of what you're seeing?

MoorPoor, 2020-10-26:
It`s hard to get it right on photo.

[image]

shawaj, 2020-10-26:
So left is a laptop screen and right is the projector?

MoorPoor, 2020-10-26:
right

shawaj, 2020-10-26:
Ok I think we may need to do a firmware update. Let me get back to you with the process shortly.

shawaj, 2020-10-26:
@MoorPoor can you provide the serial number from the box?

Also if you have any other pictures or could send a video that would be really helpful. If it's easier to email then send to sales@nebra.com

MoorPoor, 2020-10-26:
hope that is the right number: H1T003300806
the video (gif) shows another projector (bigger one) with the same test image.
[image]

ryanteck, 2020-11-05:
Hi @MoorPoor ,

We've got a few steps to try with diagnostics if possible.

Have you tried re-seating the ribbon cable that goes between the HAT and laser module?

Could you confirm what model of Raspberry Pi you are using, and how you are powering it (USB port on Pi or HAT?)

At the same time could you confirm what the specification of the power supply is that you're using. (Voltage and Ampage rating).

And finally a picture of the assembled HAT.

Thanks

MoorPoor, 2020-11-05:
Thaks for your answer!
I reattached the ribbon cable, no change
I tried Raspi Version 3B+ and 4 with 4GB Ram
and used the stock Raspberry Power supply (USB C and Micro USB: 5,1V 2,5A and a 5V, 3A to power the Pi or the HAT - no difference
[image]
[image]
[image]
[image]
[image]

ryanteck, 2020-11-05:
Thank you for doing this, we'll discuss and get back to you.

ryanteck, 2020-11-07:
Hi @MoorPoor ,

Do you have an alternative micro usb power supply that you could try, and instead try powering via the connector on the HAT rather than on the Pi itself? 

This has resolved it for another customer.

MoorPoor, 2020-11-07:
I powered the HAT with different supplys and cables also with a laboratory power supply, it says the HAT alone is using around 620mA

MoorPoor, 2020-11-07:
if one doesn't pay attention to the red tint and / or doesn't care, it may work out.
but sadly not for my use scenario
This is why I asked about the option of controlling the colors individually. That would be great.

ryanteck, 2020-11-07:
Hi @MoorPoor , thanks for doing so. I think we'll need to transfer this to email as there isn't the option to adjust induvidual colours and is most likely a fault with your unit.

I think you had a ticket open so if you could reply to that to re-open it that would be great.

ryanteck, 2020-11-20:
Hi @MoorPoor , can I just check if you re-opened the ticket via email?