NebraLtd/AnyBeam #6: Raspi 4 & ubuntu 20.10 adaption

MoorPoor, 2020-10-27:
hello.
I'm trying to get the HAT running on a Raspberry 4 with ubuntu 20.10
changing the config.txt does not work. At least I could not figure out where to put the lines. tried different versions.
This is the config.txt coming with the new version of ubuntu. In total 32 lines

```
[pi4]
max_framebuffers=2

[all]
arm_64bit=1
kernel=vmlinuz
cmdline=cmdline.txt
initramfs initrd.img followkernel

# Enable the audio output, I2C and SPI interfaces on the GPIO header
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=spi=on

# Enable the FKMS ("Fake" KMS) graphics overlay, enable the camera firmware
# and allocate 128Mb to the GPU memory
dtoverlay=vc4-fkms-v3d
gpu_mem=128
start_x=1

# Comment out the following line if the edges of the desktop appear outside
# the edges of your display
disable_overscan=1

# If you have issues with audio, you may try uncommenting the following line
# which forces the HDMI output into HDMI mode instead of DVI (which doesn't
# support audio output)
#hdmi_drive=2

# If you have a CM4, uncomment the following line to enable the USB2 outputs
# on the IO board (assuming your CM4 is plugged into such a board)
#dtoverlay=dwc2,dr_mode=host
```

ryanteck, 2020-10-27:
Hi @MoorPoor 

Typically it's added to the end of the file these lines:

```
dtoverlay=dpi24
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
framebuffer_width=1280
framebuffer_height=720
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=85
dpi_output_format=0x070027

```

tvoverbeek, 2020-10-27:
In addition you have to remove the lines:
```
dtparam=i2c_arm=on
dtparam=spi=on
```
since the display uses all gpio pins (-> no i2c and no spi)

ryanteck, 2020-10-27:
Oh yes that could be it... Thanks @tvoverbeek ...

MoorPoor, 2020-10-27:
Thanks!
In addition I had to uncomment :

ryanteck, 2020-10-27:
Excellent, glad to know that all of that fixes it.

closed 2020-10-27 by ryanteck