NebraLtd/AnyBeam #7: Reduction of speckles

ehoef, 2020-11-02:
Hi,
Love the projector and want to use it for an upcoming art exhibit.  The animation has a white background with tiny black figures.  Is it possible to reduce speckles and increase contrast (sharpness) - See image attached on how if currently looks.  I have the brightness all the way down.  Any help appreciated.

Best,
eHoef

[image]

MoorPoor, 2020-11-02:
Hei eHoef
I`m not sure about those speckel-noise but it looks like you should try to readjust the green and blue as they seem to be offset?!
I assume you checked the sharpness menu-option already? how far is the distance between projector and screen?
I also unfortunately can not get a really sharp image from a certain distance.
As the box is white you get a lot of reflections and scattered light... that also kills the contrast.
good luck with the exhibition.

ryanteck, 2020-11-02:
The speckle effect is rather hard to reduce and is a quirk of it being a laser projector. 

Our primary recommendation is to try a projector screen material as some of these reduce the effect significantly. 

you can adjust the sharpness in the OSD, are you using the HAT, developer kit or standalone?

ehoef, 2020-11-03:
Thanks all for you excellent comments.  I am using the developer kit projecting from a RPI.

- The projector is about 16 inches from the surface, might be too close by
- Did tinker with the green/blue adjustments as well as sharpness.  Distance probably mostly the culprit as well as the surface
- Have not tried projector screen material yet, it really is a tiny box, so not sure where I can get screen material in small quantities. Something to look into as well.

ryanteck, 2020-11-03:
Hi @ehoef ,

You might find a just a thin white fabric will work in place of "proper" projector screen material and is potentially something I'd try first as it can be easier to get hold of.