NebraLtd/AnyBeam #8: No OSD when pressing ENTER switch

framboise314, 2020-11-15:
Hi
I mounted the set on a Raspberry Pi 4
I installed the sh file by choosing RGB888 by default
The Raspberry Pi Boots normally, the colored screen remains on the screen forever
when I connect to the Pi4 with VNC I can use it normally, open terminal, web browser...
I checked the correct connection of the ribbon cable on Nebra HAT

The LASER projector turns on normally and displays the NEBRA ANYBEAM splash screen as shown in the joined picture 
However, when I press the ENTER button (more than 10 seconds) I don't get the OSD menu display.
As for the LEDs :
- Indicator LED 1 is permanently lit (normal operation)
- and an LED flashes under the radiator of the laser block

Do you have an idea of what could happen, where to search  and how to remedy this defect? 
How can I verify that the HAT works ?
regards
francois
[image]

ryanteck, 2020-11-15:
Hi @framboise314 ,

Just to check that you're using Raspbian?

Could you post a couple of screenshots or attach the ```/boot/config.txt``` file here to help confirm that part has been setup correctly by the installer script.

Thanks

ryanteck, 2020-11-16:
Hi @framboise314 , I was wondering if you saw the above response to help us diagnose this issue?

framboise314, 2020-11-17:
Hi Ryan @ryanteck 
yes i use Raspbian for the tests and no sorry I was pretty busy with [the release of my new book](https://www.editions-eni.fr/livre/scratch-et-raspberry-pi-projets-maker-pour-s-initier-a-l-electronique-et-a-la-robotique-2e-edition-9782409027901) and I hadn't seen the message. Your request about config.txt has clicked! So as I had installed the sh script on a card that had already been used quite a bit I redid the SD card and updated the Raspberry Pi OS. I installed the anybeaminstall script again. This time everything went well. The colored display disappears on the screen and the Raspberry Pi OS wallpaper is well displayed on the screen. 
On the LASER projector the AnyBeam splash screen disappears and the RasPiOS desktop screen is displayed. And this time **the ENTER button works well** and I can configure the projector with the menu.
I think the advice to give to users is to **start with a freshly installed SD card**...
thank you for pointing me to this solution.
have a nice day 
francois

closed 2020-11-17 by framboise314

ryanteck, 2020-11-17:
Excellent to hear it's fixed. Congrats on your book launch.