NebraLtd/AnyBeam #10: Few questions

alexspace360, 2020-11-26:
1- official website ? 
2- official Email ? 
3- can i project at 40 cm away without losing details ?

ryanteck, 2020-11-26:
Hi @alexspace360 ,

1 - Our website for selling the units is at https://uk.pi-supply.com/products/nebra-anybeam?, https://uk.pi-supply.com/products/nebra-anybeam-developer-kit & https://uk.pi-supply.com/products/nebra-anybeam-hat-laser-projector-for-the-raspberry-pi

2 - You can email us on sales@pi-supply.com

3 - 40CM would be fine but likely require content that's big for it to be clear. Typically we recommend from 1M distance but it can be closer.

ryanteck, 2020-11-27:
Hi @alexspace360 ,

We've taken a couple of images showing a video playing at about a 40CM distance. I hope these help.

[image]
[image]
[image]

closed 2020-11-27 by ryanteck