NebraLtd/AnyBeam #11: Raspberry Pi 0 W - no image

jahevos, 2020-11-30:
I cant get the image from my raspberry pi 0w dpi_video out. Basically I am stuck with the nebra startup screen. I tried this under a number of distros - libreelec,kodi,raspian and raspberry os. I get, however, a proper image when setting it up with a rpi 2.
I tried all of the suggested config methods from the user guide but to no avail.
Also no matter which power supply and usb wire I use I do get a yellow flash in the top right corner. 
Any Input would be appreciated - maybe anyone else ran into these kind of problems.

cheers 
J

ryanteck, 2020-11-30:
Hi @jahevos ,

If it's where it works with your Raspberry Pi 2 fine with the same SD Card image as you're trying on your Zero I would suspect it could be a faulty Pi Zero W.

Did you buy one with the header pre-soldered or did you solder it yourself? It could be one of the joints isn't quite good enough for the signals required for the HAT.

jahevos, 2020-11-30:
Hi Ryan thanks for your quick reply.
I got a pre-soldered pi, so it should be okay but I will look into it later. Cheers

ryanteck, 2020-11-30:
That *should* be fine then. If you could take a picture of the header I can see if there's any visual defect with it but it would be odd if it's pre-soldered.

jahevos, 2020-11-30:
Hi Ryan,

I gave everything another go, with no success. I'll get another pi zero and see what happens. I'll keep you posted on the outcome. Best j

jahevos, 2021-02-28:
Sorry for the late answer. Fault was defective rpi0 -the new one resolved the issue thanks

closed 2021-02-28 by jahevos