NebraLtd/AnyBeam #14: Temporally moving pattern

beniroquai, 2020-12-07:
We are working with the HAT on a microscope and observe some wiggeling of the pattern. We recorded a static pattern of single white points on a black background and capture it at ~60Hz. You can see that the pixels will vary spatially over time:
[image]

Are you aware of this? Is there any chance to change that (e.g. different power supply?) For Video projection it probably doesn't matter since it is a variation in sub-degree range..

Thanks a lot!

beniroquai, 2020-12-16:
Hey, some further analysis suggests, that it is also visible on the wall:
[image]
We basically addressed some vertical stripes fulscreen using pygame on the raspberry pi. 
You see the bouncing or variation of the thickness. It happens also with vertically addressed stripes. There are regions where it is more severe:
[image]

It also seems that the RGB channels are not inline. Unfortunately, the buttons won't open the menu for some reason in order to correct for that. Any idea? 

[image]

Thanks a lot!

ak1394, 2020-12-22:
I had the same issue where I couldn't open the menu when unit was connected to HDMI source (I have the projector version, not the HAT) but it worked for me with only the power connected, no HDMI.

beniroquai, 2020-12-28:
Unfortunately, this didn't help. We tried unplugging all external device without any effect. 
I wonder if the hitter is a known problem from this Ultimems MEMS?