NebraLtd/AnyBeam #16: AnyBeam HAT to OE Sensor FPC Connector/Housing Specs Please.

pixma, 2020-12-15:
Can I have the spec of the FPC connector where the FPC Cable is used to connect the HAT and the Mems Lase Sensor Module?
If yes, I would like to know more about the connector's
- Pitch?
- Top Contact or Bottom Contact?
- Positions: 30 | OK
- Voltage Rating

and so on and so forth, of the FPC Connector. So we can choose the FPC Connector correctly to take in the FPC Cable which comes out from OE Mems Sensor Module Itself.

Regards,

shawaj, 2020-12-16:
[image]
Pinout diagram for it.

pixma, 2020-12-16:
Hi,

Thank you @shawaj for the pinout description, but, I am looking for something like this, if you can confirm if it is 5mm pitch bottom contact FPC Connector. Or if the PFC connector has some other spec?

[image]

shawaj, 2020-12-16:
@pixma ah yes sorry should have said I'll have that detail for you soon

pixma, 2020-12-23:
Hi,

I am looking forward to that info.
It's important info/ Hence, this is a gentle reminder @shawaj or anyone who have this info, can you please share when you can.

Regards

shawaj, 2020-12-24:
[image]

So it's 30 pin and 0.5mm pitch

shawaj, 2020-12-24:
@pixma was that what you needed?

pixma, 2020-12-24:
Thank you @shawaj , yes this information is good enough to select the mating connector.

Regards.

closed 2020-12-24 by pixma