NebraLtd/AnyBeam #17: Is it possible to control the project settings programmatically?

MrNHQ, 2021-01-07:
Hello,

I would like to control the projector setting programmatically from a raspberry pi. 
One option, using the anybeam HAT, could be "hacking the buttons" with transistors  and resistors and connect them to an arduino. Finally, connect the arduino to the raspberry pi so that raspberry pi is the controller. 
I don't know if it is possible with the other type of anybeam (developer kit or production)?

Are there other easier alternatives? 

Note, I do not have the anybeam HAT yet, it was out of stock. I contacted distrelec and they will receive new pieces in a week or two :).

GregFinger, 2021-03-23:
I'm also interested in this

beniroquai, 2022-10-11:
Das kann ich machen. Bis wann braucht ihr die Zeichnungen?
Im besten Fall kann ich es morgen anfertigen. Habt ihr eine Maßgabe
hinsichtlich der Radii?

Viele Grüße
Benedict

Am Di., 26. Apr. 2022 um 14:09 Uhr schrieb ***@***.***>:

> Liebes UC2 Team,
>
>
>
> aus Sicherheitsgesichtspunkten bitte ich um eine Änderung der
> Laserzuschnitteile aus 3mm  Acryl.
>
>
>
> Grundsätzlich sollten alle Teile an den Außenkanten einen Radius haben und
> keine rechtwinkeligen Kanten. Das kann einfach zu Verletzungen führen.
>
>
>
> Könnt ihr das noch machen? Oder soll LABmaker für die Fertigungsserie alle
> Schnitteile eigenständig neu erstellen?
>
>
>
> Viele Grüße
>
> York
>
>
>
>
>
>
>
>
>
>
>
>
>
>