NebraLtd/AnyBeam #19: No OSD when powered via a computer USB port

jampekka, 2021-02-10:
My unit (devkit) does not show the OSD when the unit is powered via a computer USB port. When powering via a wall charger it works. Is this by design?

The chip also seems to have a STM32 controller that sets up a virtual com port over USB. Is it possible to hack the controller, and is there any documentation for it available?

ryanteck, 2021-02-10:
@jampekka It's most likely that the computer isn't able to provide enough power to the unit which is cauisng the abnormality.

The Virtual Com port is only to update the firmware if it is required and not usable for nothing else.

jampekka, 2021-02-10:
@ryanteck May be, but I doubt it. I can speed charge phones with that USB port.

For anybody interested, it seems that it normally boots in STM32 Virtual ComPort mode (showing as /dev/ttyACM* in linux). Didn't manage to get it to say anything yet through that port though.

If you hold the enter button when plugging power, it enters into DFU compatible bootloader mode. I was able to dump the settings and the firmware using dfu-util.

I gather Nebra is not intending to support hacking of the device of any kind?