NebraLtd/AnyBeam #22: Can anybeam HAT project structure light for 3D scanning

leoteo1, 2021-04-23:
I am looking for a high brightness development projector for 3D structure light scanning. can anybeam HAT project struture light (sine pattern) and synchronize (vertical sync) with cameras for frame trigger?

LutzKoepke, 2021-05-09:
I would be interested in that too. I would like to use the anybeam HAT as a scanner to determine the dependence of the sensitivity of a photonic device (a large diameter photomultiplier) as function of its surface coordinate. Since I do not necessarily want to use a high speed FADC data acquisition (such as a RedPitaya) or oscilloscope for readout, it would be good to be able to lower the "scanning speed", i.e. the frame rate, by a lot. In principle, the hdmi settings in config.txt could be adapted by adding something like  hdmi_cvt 900 720 1 4 for a frame rate of 1/s and a 5:4 screen. Would something like this be possible or does the hardware prevent to lower the framerate? I read that one can configure the hat such that one can salvage an i2c connection but no SPI. A really fast ADC (i.e. a few MHz sampling) can thus not be attached, however, there are i2c ADCs that can work with close to 200 kHz ... Of course it would be even better if one could address the Ulitmens sensor via the RPI hat directly, without going through the hdmi protocol ...

leoteo1, 2021-05-16:
> I would be interested in that too. I would like to use the anybeam HAT as a scanner to determine the dependence of the sensitivity of a photonic device (a large diameter photomultiplier) as function [...]

leoteo1, 2021-05-16:
LutzKoepke, did you find out any other way to project structure patterns using project?

closed 2021-05-16 by leoteo1

LutzKoepke, 2021-05-16:
dear leoteo1, 

I admit that I have not yet bought the anybeam HAT so that I could do some "reverse engineering". Since you do not seem to need 3 colours and light intensity modulation, you might be better off with solutions by other vendors that offer MEMS devices that you can address directly. The anybeam solution has the advantage of price and directly coupling to the RPI. Unfortunately, I could not find specs for the Ultimems sensors  http://www.ultimems.com/technologydetail/index.html#t2 , which come in 3 mirror sizes http://www.ultimems.com/en/index.html#solutions . From the pictures one sees that the mirror has 6 ? connections plus whatever is needed to drive the 3 lasers. A raspberry pi has 40 pins of which 8 are connected to ground and 4 are connected to 5V of 3.3 V power. That gives you 28 addressable pins. The anybeam hat seems to use all pins in mode 1 [https://github.com/NebraLtd/AnyBeam/tree/master/Hardware:  _we use every pin availble on the Raspberry Pi. That means you don't get the standard users i2c pins, the UART(Tx/Rx)pins, the EEPROM i2c pins and the hardware SPI pins.]__ . 24 pins are used as a parallel interface to provide 8 intensity bits for each color. I assume that ther are 4 additional pins for defining the mirror action. In mode 4, 18 bits are used for addressing the lasers, and GPIO 26/27 are freed for an I2C bus. In the latter case, there should be  4 additional bits available for free use; I am not sure why no SPI bus option is provided, which would allow for attaching faster ADCs. 

I have still not decided to buy the anybeam hit in order to understand the connections using an oscilloscope. In particular I would need to find out what the 4 pins do that address the mirror action and which pins these are. For now I only bought the Seeed Studio Grove - I2C ADC 12-bit ADC121C021 fast ADC (185000 samples/s) in order to see whether I can actually reach that readout speed. 

I may have a more careful look at the pictures of the hat (e.g. https://user-images.githubusercontent.com/1878314/57315703-06315580-70ec-11e9-8bf4-dd6b58aac2cf.jpg ), maybe one could guess from those which pins are used for what?

LutzKoepke, 2021-05-16:
addendum: anybeam probably uses the "larallel display interface" : https://www.raspberrypi.org/documentation/hardware/raspberrypi/dpi/  _which allows for parallel RGB displays  either in RGB24 (8 bits for red, green and blue) or RGB666 (6 bits per colour) or RGB565 (5 bits red, 6 green, and 5 blue)._