NebraLtd/AnyBeam #23: 3D model / case ?

MoorPoor, 2021-07-02:
Hei there...
I wonder if there is a 3D model of the HAT existing and / or a 3D case for printing?
finally I would like to use the thing but would like to give it a housing.
greetings!

MoorPoor, 2021-07-02:
As I could not find any Raspberry Case, I did that one here... a bit dirty, but will hopefully meet my requirements...
maybe helpful to someone.
 https://www.tinkercad.com/things/k1F4rvU07ks 
 based on that one: https://www.thingiverse.com/thing:3361218

jahevos, 2022-03-22:
Hey, nice design, would it be possible to upload the stl files for top and bottom. thx

MoorPoor, 2022-03-22:
[Nebra AnyBeam Raspi3 Case.zip](https://github.com/NebraLtd/AnyBeam/files/8324670/Nebra.AnyBeam.Raspi3.Case.zip)
Here you go. However, I only have the version for the Raspberry 3. I hope it helps.

jahevos, 2022-03-22:
Awesome thanks