NebraLtd/AnyBeam #28: Raspi HAT "waiting for connection"

MoorPoor, 2022-03-31:
Hei.
finally I found a good usecase for the RaspiHAT.
The moment everything works it asks me to connect a device.
The display comes pretty much immediately after plugging in the power source.
I have never seen this in all the many past boots.

is it possible that the HAT needs to have its firmware reloaded in some way?
before that it ran successfully with the Raspi 4
I tried it again with a Raspi3, without success

[image]

ChristopherRush, 2022-04-01:
@MoorPoor I'm assuming you did run the install script right? Check everything was configured in the config.txt file.

MoorPoor, 2022-04-01:
no, I configured the config file by hand, and it was running and projecting well.
but suddenly no longer after a restart. without any changes