NebraLtd/AnyBeam #31: MIPI/DSI interface specs

SinanAkkoyun, 2023-04-10:
Hello,

I'd like to use the projector module without the lontium pi adapter board.
What specs and timings does the DSI interface have? (if the FPC is even DSI, or is it proprietary?)

Thank you

kfl02, 2023-11-29:
Hi,

I can't answer your question, but did you by any chance get the stock Anybeam hat working on a raspberry pi 4 with the latest Raspberry OS?
The dpi24 overlay is deprecated and the new vc4-kms-dpi-generic overlay doesn't work out of the box for me.

I seem to need timing values as well (got some flickering out of the nebra with some randomly chosen values) and I also noticed the lontium chip on the board. I am an absolute layman with these kind of things.

Best
Karsten

SinanAkkoyun, 2023-11-30:
Yes, I got the Pi HAT and I gave up

I'm closing this issue as this will never resolve, frel free to open another one if you are trying to make it work

Best Wishes and good luck!

closed 2023-11-30 by SinanAkkoyun